patrykcieslak/stonefish #11: Compiling on MacOSX

jdbodyfelt, 2020-07-22:
Builds great in Ubuntu 20.04 (work remotes), however my laptop runs Mac OSX Catalina. Reckoned I'd post this small bug fix for others that might hit this same problem.

**Describe the bug**
Following Unix install instructions, there path errors within MacOSX `SDL2` and `glm` libraries. 

**Solution**
1. Install both SDL2 and glm using `brew install <package>`. 
2. Use `brew info` to determine install paths. 
3. Make a symbolic link within `Library/include`. 

For example, my build succeeded with the following two linkages:
* `ln -s /usr/local/Cellar/glm/0.9.9.8/include/glm ../Library/include/glm`
* `ln -s /usr/local/Cellar/sdl2/2.0.12_1/include/SDL2 ../Library/include/SDL2`

If you're using Windows...may Heaven have mercy on your poor masochistic soul.

jdbodyfelt, 2020-07-22:
ACKKK. Spoke too soon - builds fine, but execution fails with `[CRITICAL] SDL2: Failed creating OpenGL context at version requested`...

patrykcieslak, 2020-07-22:
Hello!
I have bad news for you. I am also a Mac user and I was developing the library on Mac when I started, but I work on Ubuntu now and we always combine the simulator with ROS. I had to made a decision and move to OpenGL 4.3, to be able to use compute shaders. Apple is supporting max OpenGL 4.1 and that is why SDL2 fails. They don't care about the OpenGL anymore and 4.1 is an ancient version. I don't want to be limited. Therefore, I dropped the compatibility with Mac. 
I will think about Windows users because as much as I hate this system, there is a lot of poor fellow researchers using it :)
Cheers

closed 2020-07-22 by patrykcieslak