patrykcieslak/stonefish #12: Try get the full images of ColorCamera and FLS

martingallegos11, 2020-07-25:
**Is your feature request related to a problem?**
Hi! I can't correctly export ColorCamera and FLS images for use in an external dependency (LCM). I don't really know the 'correct' way of getting the image, and it would be awesome if you could help me with that.

**Describe the solution you'd like**
I would love an easy way of getting the full images correctly, of the ColorCamera and FLS so I can later use them in any way I see fit.

**Describe alternatives you've considered**
I've considered using the function *getImagDataPointer()* of both sensors, like this:
```
    //ColorCamera
    uint8_t* image = (uint8_t*)camera-> getImageDataPointer();
    unsigned int res_x, res_y;
    camera-> getResolution(res_x, res_y);
```
```
    //FLS
    auto im = (float*) fls-> getImageDataPointer();
    unsigned int nRanges, nBeams;
    fls-> getResolution(nBeams, nRanges);
    glm::vec2 range = fls-> getRangeLimits();
```
I believe this is not the correct way, because later I'm having some segmentation faults related to that. I believe that in retrieving the image I'm stepping on forbidden memory space, but I'm not really sure. It would be awesome if you can point me in the right direction.

patrykcieslak, 2020-07-27:
Hello! 
One of the ways to understand how to get the data from the sensors correctly is to look into the code of the stonefish_ros package. Even if you are not using ROS it shows how to subclass the standard simulator, read sensors, update actuators and so on.
The thing is that when you create any sensor (in BuildScenario()) that is a subclass of the VisionSensor class, i.e., sonars and cameras, you need to set a handle to a callback function that will be called when the image is ready to download from the GPU, e.g.,
```
sf::ColorCamera* cam = new sf::ColorCamera(....);
cam->InstallNewDataHandler(std::bind(&MySimulationManager::MyCallbackName, sim, std::placeholders::_1));
```
where sim is a pointer to the MySimulationManager and the callback looks something like this:
```
void MySimulationManager::MyCallbackName(ColorCamera* cam)
{
    uint8_t* image = (uint8_t*)cam->getImageDataPointer();
    //Do something with the image data
}
```
It is important to remember that this callback is not asynchronous and it makes the simulation wait for it to return. It is done like this because there is no guarantee that the image will be still ok if you access it later. There could be double-buffering on the side of the simulation but because the image is then sent as ROS messages in my case I wanted to avoid another copy.
Therefore, in this callback you should only `memcpy()` the data from this image pointer to your own buffer and exit. Otherwise you will be freezing the simulation every time you do some processing in this callback, so every time a new image is ready.

I am starting to work on a documentation finally but as you can imagine it is the least interesting part ;)
Cheers

martingallegos11, 2020-07-27:
Thanks very much for you help! It will be very usefull.
Cheers

closed 2020-08-20 by patrykcieslak