patrykcieslak/stonefish #15: Problem running tests ?

felipecode, 2021-01-05:
Hello Patryk !
Thanks a lot for this simulator. Seems really cool but I am having a bit of problems
getting it to work. Might be something really dumb I am doing since I am bit rusty on C++ ( Getting back to it now) 
**Describe the bug**

I was able to compile and install the library without any problems.
But for some reason I cant compile the test examples, and when I make my own example following your tutorial I have some linking problems.
It might be my compiler version or something like that but i feel i don't have the info to fix it.

### The test examples.

First, compiling then directly with the provided cmake does not work.
doing    
```
     cd Tests
     mkdir build && cd build && cmake .. && make 
```
gives me include error, since the include does contains the folder Stonefish in the path.

```
/home/felipe/stonefish/Tests/FloatingTest/main.cpp:26:10: fatal error: core/GraphicalSimulationApp.h: No such file or directory
 #include <core/GraphicalSimulationApp.h>
```
fixing that gives me an error on the data folder macro

```
/home/felipe/stonefish/Tests/FallingTest/main.cpp:49:35: error: expected primary-expression before ‘(’ token
     FallingTestApp app(std::string(DATA_DIR_PATH), s, h, simulationManager);
```

### My own example

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:  ubuntu 18.04
 - Compiler - g++ 8.4

**Additional context**
Add any other context about the problem here.

patrykcieslak, 2021-01-07:
Hola Felipe!
Nice to hear from you and see you interested in my project.

The examples that are in the Stonefish library are prepared just to make it easy for me to test things during development. 
If you want to play with them you need to run cmake -DBUILD_TESTS=ON .. (assuming you created the build directory under the main directory).
This will compile and link the library and the examples locally, without installing the lib in the system.Then you can launch any of the examples in the Tests directory.

If you want to create your own example, using the shared library installed system-wide, you need to link a few libs.
The easiest way is to use pkg-config. You can also look into the /usr/local/lib/pkgconfig/stonefish.pc file. You will see the flags needed for the compiler and linker. 

I will add some compilation instructions. Good that you noticed it.

Patryk

felipecode, 2021-01-07:
Thanks ! It works. I am a bit rusty in c++ indeed ! The compilation instructions would be great.

closed 2021-01-07 by felipecode