patrykcieslak/stonefish #16: Rudders

nilsbore, 2021-01-07:
**Is your feature request related to a problem? Please describe.**
I am not sure if this is a bug or a feature.

I am trying to get rudders working in our stonefish simulation.
I remember you suggesting to simply add a link with a rudder mesh model
and attach it to a revolute joint that can be used to control the rudder angle.
I tried that but I have some problems getting the simulation to be stable with
that kind of actuator. What typically happens either directly when you start
up the simulation or when you start to control the rudder is that the vehicle
"flies" away. It seems that somehow too big a force is applied
by the joint controller.

I have tried various combinations of rudder mass, `max_torque`, `position_gain` and `velocity_gain`.
Do you have any ideas of combinations that you think should be more stable
(although maybe not physically realistic)?

**Describe the solution you'd like**
The rudder joint turning to the setpoint, without the vehicle flying away :).

**Describe alternatives you've considered**
Implement a dedicated rudder actuator.

Thank you!
Nils

patrykcieslak, 2021-01-07:
Hello!
Hard to say what is happening without seeing the geometry you are using. 
I think modifying these three variables may help but the vehicle should not be flying anywhere anyway.
Can you make some video?

nilsbore, 2021-01-07:
The rudder mesh is available [here](https://github.com/nilsbore/lolo_stonefish_sim/blob/noetic-devel/data/lolo_auv/mesh/fin.obj) and the complete vehicle definition is [here](https://github.com/nilsbore/lolo_stonefish_sim/blob/noetic-devel/data/lolo_auv/lolo_auv.xml) with the relevant lines start from [here](https://github.com/nilsbore/lolo_stonefish_sim/blob/noetic-devel/data/lolo_auv/lolo_auv.xml#L53). I've captured a video available [here](https://drive.google.com/file/d/1QIunb4DjyLSJKZ6AQwkmCTR7IBrb9RgL/view?usp=sharing) but the screen capture software makes the computer pretty slow. What you can see is that it disappears from view as soon as I give a command to the vertical rudders (the ones whose mesh I linked). But what I have observed is basically the joint actuators seem to go back and forth very rapidly while the vehicle is flying away.

nilsbore, 2021-01-13:
@patrykcieslak [This movie](https://drive.google.com/file/d/1QIunb4DjyLSJKZ6AQwkmCTR7IBrb9RgL/view?usp=sharing) that actually follows the vehicle may give a better idea of what's happening. Again, after I change the position of the vertical rudders, the vehicle flies away.

nilsbore, 2021-01-25:
@patrykcieslak Do you have any idea what might cause this issue? I would like to try and solve it but I don't know where to start.

patrykcieslak, 2021-01-26:
I was experimenting a bit with your vehicle directly in Stonefish, without ROS, and I can see these problems although it seems more stable and I can control the rudders without exploding the system. The problem is that they just don't work as they should. I think that when they are locked in some position the computed drag is to low to change the direction of the AUV. the effect is that you cannot really steer the AUV with the rudders :( In real world they should be producing lift forces and this is not simulated. I was wanting to implement a Fin actuator which would be, like the Thruster actuator, based on some model, applying forces calculated from parameters of the profile and angle of attack. The graphical representation would be just for visualisation like with the propeller. I have seen somewhere that for aerodynamic profiles (NACA) it is possible to develop some theoretical formula that will give us lift and drag vs angle of attack. I have no time now to focus on this because I am developing things related to the projects we run here, but at some point I will work on it. You're welcome to explore the topic ;) I think this is the only manner to make it work and behave realistically.

nilsbore, 2021-01-27:
Thank you for investigating and for the detailed response! I will get the vehicle working using differential thrust for the while being. And I may try to get a simplified rudder model working. I suspect the system exploding in this case is due to my home laptop being slow. Do you know if I could tune any of the settings to make the physics simulation more stable on low-powered computers?

nilsbore, 2021-01-29:
Added rudder support in https://github.com/patrykcieslak/stonefish/pull/17 and https://github.com/patrykcieslak/stonefish_ros/pull/13

closed 2021-06-14 by patrykcieslak