patrykcieslak/stonefish #21: Simulated side scan sonar looks unexpected

luxiya01, 2021-05-07:
**Describe the bug**
When trying to simulate side scan sonar image using stonefish according to [the documentation](https://stonefish.readthedocs.io/en/stable/sensors.html#side-scan-sonar-sss), the resulting simulated sensor output appears to be somewhat erroneous. 

Specifically, it looks like that the port and starboard channels are exact complements of one another. The below SSS screenshot was taken when a simulated robot travels above a flat seafloor with constant velocity.

**Screenshots**

[image]

patrykcieslak, 2021-05-24:
Hello! It indeed looks strange. I was not playing much with the sonars yet so I al happy to see someone is testing it. To me it looks like the vehicle was having some sudden roll change periodically. Is it moving in a stable way?

luxiya01, 2021-05-25:
Hi! Thanks for the reply :)
The vehicle was running a Neptus mission at constant velocity with waypoints on the same height, so I believe it was moving stably forward. 

Here is another image when the vehicle is semi-stationary, i.e. it was slowly drifting with the current. When zoomed in to the details, the same pattern observed is visible: the port and starboard channels seem to complement one another, when one side indent, the other side extrude. It is almost as if one single camera image is split into two sections.
[image]

patrykcieslak, 2021-05-25:
Please record the pose of the robot in time (position and orientation), because to me it still seems like the robot was just oscillating in roll. When we see the motion we will be sure it is not this issue.

patrykcieslak, 2023-02-03:
Surface waves are not really doing anything if you don't set them up to be actual geometrical waves, which is an option but still considered WIP.  What is happening in your case is something related to the model setup. First of all check where is your buoyancy center and where is the gravity center. You need to set up the model so that heavy parts are at the bottom and buoyancy at the top, like in the real vehicle. Even if your vehicle is a cylinder it has some parts inside and normally I just model these parts in a simplified way and give them masses so that the simulator can compute the actual gravity center itself.