patrykcieslak/stonefish #22: Thruster model error?

nistenius, 2021-05-23:
**Describe the bug**
I'm experiencing a very odd behaviour with the thruster models. Regardless of rpm value 0 - 1000, the vehicle moves with approximately the same velocity. When changing the thruster coefficient, the velocity changes but the same behaviour with a constant speed, regardless of rpm value, is still there.

I'm running the simulator within the SMaRC ROS framework: 
https://github.com/smarc-project

I have tried to vary coefficients and max rpm settings but it does not change this behaviour.

Any thoughts on what this could be related to?

patrykcieslak, 2021-05-24:
Hello! Please check what you are sending to the thrusters. The set point is in a range from -1 to 1, the rpm is controlled linearly between 0 (stopped) and 1 (max rpm).

nistenius, 2021-05-24:
Thanks Patryk for quick reply,

Yes, that’s most likely the issue. We’ll update and get back to you regarding how it goes.

Best regards
/Ivan

closed 2021-06-14 by patrykcieslak