patrykcieslak/stonefish #25: Graphics free mode?

silent-code, 2021-12-10:
**Describe the bug**
No bug. Just a question.

How can one run the stonefish simulator in a mode with no graphics displayed? 

Thanks for the awesome simulation btw!

patrykcieslak, 2021-12-13:
Hello! 
You're welcome hope you find it useful.

I am not sure what does "no graphics" mean to you. You just don't want to see the Stonefish window (this is not really possible) or you want to run truly in console mode (no graphics initialised, no cameras and sonars available.
If you want the console mode simulation which basically gives you dynamics and sensors not based on graphics, then please take a look at the "stonefish_ros" repository. You will see there an example of a simulator without GUI/Graphics.

Cheers!

silent-code, 2021-12-13:
Thanks for the response. I'll look at that.

closed 2021-12-13 by silent-code