patrykcieslak/stonefish #26: Simulation Reset

silent-code, 2021-12-22:
**Is your feature request related to a problem? Please describe.**
More of a question than a request. I'm doing reinforcement learning which requires episodic trajectory rollouts. It would be convenient if there were a way to reset the state of the robot inside the sim and not have to restart the app everytime a new episode rollout is required or drive the robot back to it's starting location and pose. Is there anyway to reset the robot state which would include being able to reset for example x, y, z, vx, vy, vz, roll angle, pitch angle, yaw angle, omega, accel, etc.? 

I guess it would look like teleportation in the sim haha!

**Describe the solution you'd like**
State reset as a function call, perhaps.

bastianfeliz, 2022-02-28:
Hello @silent-code, I'm working on the simulator and I think we both are trying to achieve the same functionality, though with different ends. Up until now I've managed to move my robot using a `SolidEntity` method called `setCGTransform` inside the `SimulationStepCompleted` function. Here is a cut of the code that does that:
```        
sf::Robot* robot_ptr = getRobot("robot_name");
sf::SolidEntity* roventity = robot_ptr -> getBaseLink();
roventity -> setCGTransform(CGTransform);
```
This achieves to move the robot during the simulation step, however, in the next step the robot goes back to where it was and its velocity changes in a wierd way that I don't understand yet. I'm currently trying to better understand the code to find a solution but I'm still a novice to C++ haha. If you have worked out a way to do this or if you want more details let me know! Also, @patrykcieslak if you can help out with this I'd be really grateful.

silent-code, 2022-03-07:
Have you tried creating a thruster for the robot and then simply setting the thruster value? This will give you smooth steady motion during each simulation step without unwanted velocity changes. For example in the TestManager::BuildScenario() function:

sf::Thruster* thSurge = new sf::Thruster("motor_name", prop2, 0.1, std::make_pair(.64, .64), .5, 3500.0, false, false, sf::Transform(sf::Quaternion(0,0,0), sf::Vector3(0,0,0)); 

and then in the SimulationStepCompleted function:

sf::Thruster* th5 = (sf::Thruster*)getRobot("robot_name")->getActuator("motor_name");
th5->setSetpoint(.8);

patrykcieslak, 2023-02-03:
Hello guys!
If you want to completely restart the simulation there is a method for this. I had it implemented under the SPACE key but I commented it out. You basically need to stop the simulation with `StopSimulation()` from the `SimulationApp` class and then you have to call `RestartScenario() `from `SimulationManager` class and then `StartSimulation()` from the `SimulationApp` class again. Depending where you do the calls you need to use the right objects to call the methods on.
The result will be a complete reset of all objects in the scene because the `RestartScenario()` is actually calling `BuildScenario()`.

silent-code, 2023-02-03:
Awesome thank you!

closed 2023-04-26 by patrykcieslak

bastianfeliz, 2023-11-30:
@silent-code Hello! Did you manage to restart the simulation in the end? I've given it some shots with Patrick's answer but I am running into segmentation faults. From the info I've gathered using GDB, I believe the simulator app keeps rendering objects while the manager is restarting the scenario and maybe some things are getting deleted ahead of time. I'll keep trying to debug my implementation, but if you have any information you can share I'd be very grateful.

Cheers!