patrykcieslak/stonefish #27: Drag computation for compound hull

silent-code, 2022-02-11:
On line 324 of Compound.cpp, it states that drag is computed for externally declared parts that are submerged. That makes sense. However, does the app compute drag  for the hull, for example, in the following declaration of the compound vehicle hull since it's not explicitly labeled as 'external'? 

//Build whole body
 sf::Compound* vehicle = new sf::Compound("Vehicle", phy, hullB, sf::I4());

silent-code, 2022-11-04:
@patrykcieslak: Hi Patrick, can you clarify this matter? Just wondering if the above declaration gets included in the drag computation since it's not explicitly labeled external?

Thanks,
Tim

patrykcieslak, 2022-11-04:
Hello! If you looked into the header file you would see that the part that is passed to the constructor is called "firstExternalPart". My thinking was like: if there is some physical body made of parts, at least one is gonna be external, so let's force that the first one passed is external. Actually, when I think about it now, it may be unnecessarily restrictive, because maybe you want to combine some internal parts into one and then declare the external ones separately...
Sorry for not responding earlier!

silent-code, 2022-11-04:
Thanks Patrick! I missed that in the header!

closed 2022-11-04 by silent-code