patrykcieslak/stonefish #28: Building fails with compilation error on AcousticModem.cpp

EvenStrangest, 2022-06-01:
Changing 'CMAKE_CXX_STANDARD' to '20'  changed nothing.

`[  4%] Building CXX object CMakeFiles/Stonefish.dir/Library/src/comms/AcousticModem.cpp.o
/mnt/c/Users/asafe/BoatSpeed/stonefish/Library/src/comms/AcousticModem.cpp: In static member function ‘static sf::AcousticModem* sf::AcousticModem::getNode(uint64_t)’:
/mnt/c/Users/asafe/BoatSpeed/stonefish/Library/src/comms/AcousticModem.cpp:72:34: error: expected unqualified-id before ‘&’ token
   72 |     catch(const std::out_of_range& oor)
      |                                  ^
/mnt/c/Users/asafe/BoatSpeed/stonefish/Library/src/comms/AcousticModem.cpp:72:34: error: expected ‘)’ before ‘&’ token
   72 |     catch(const std::out_of_range& oor)
      |          ~                       ^
      |                                  )
/mnt/c/Users/asafe/BoatSpeed/stonefish/Library/src/comms/AcousticModem.cpp:72:34: error: expected ‘{’ before ‘&’ token
/mnt/c/Users/asafe/BoatSpeed/stonefish/Library/src/comms/AcousticModem.cpp:72:36: error: ‘oor’ was not declared in this scope
   72 |     catch(const std::out_of_range& oor)
      |                                    ^~~
make[2]: *** [CMakeFiles/Stonefish.dir/build.make:230: CMakeFiles/Stonefish.dir/Library/src/comms/AcousticModem.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/Stonefish.dir/all] Error 2
make: *** [Makefile:136: all] Error 2`

EvenStrangest, 2022-06-01:
Solved by including 'stdexcept', as per [[Exception 'out_of_range' not member of std?](https://stackoverflow.com/questions/29607544/exception-out-of-range-not-member-of-std)](https://stackoverflow.com/a/29607575).

patrykcieslak, 2022-09-05:
Thanks for reporting. Should be fine now.

closed 2022-09-05 by patrykcieslak