patrykcieslak/stonefish #31: range and intensity data for Mechanical scanning imaging sonar (MSIS)

linzha0, 2023-01-06:
**Is your feature request related to a problem? Please describe.**
This is not related to a problem.

**Describe the solution you'd like**
MSIS sensor publishes another data format such as [sensor_msgs/LaserScan](http://docs.ros.org/en/melodic/api/sensor_msgs/html/index-msg.html)

**Additional context**
Thanks for sharing this awesome project. We are trying to get range and intensity from the MSIS sensor. I checked the source file generated sonar image such as the _OpenGLMSIS.cpp_ , but this seems strongly related to OpenGL, which I am not familiar with.

Could you give some suggestions to get raw range and intensity data from the MSIS sonar? Maye some tips to extract data from the OpenGL buffer, or maybe post-process the sonar image to get the data we want. 

Thank you so much !!

patrykcieslak, 2023-02-03:
Sonar as you noticed is implemented using OpenGL and the output is an image that is closely related to the real sonar data. This data is representing in X the beams of the sonar and in Y the distance bins, while the value of each pixel represent the echo intensity. What you would like to achieve is a profiler sonar and this can be done by post-processing the image using some threshold for the echo intensity.

linzha0, 2023-06-13:
I currently found `getCurrentRotationStep` function from MSIS, so we can get the current measurement from the raw image(bin-beam format), and get raw range and intensity for each bin-beam.