patrykcieslak/stonefish #32: Is it possible to use Stonefish for surface vehicles/boats?

therishidesai, 2023-01-25:
I want to be able to use the stonefish simulator for boats and I was wondering if that has ever been tried.

silent-code, 2023-01-30:
Yes, there is a minimal example in the tests folder: https://github.com/patrykcieslak/stonefish/blob/master/Tests/FloatingTest/FloatingTestManager.cpp

patrykcieslak, 2023-02-03:
Yes it is possible, the buoyancy is working realistically, so if you build a boat it will not sink. However, I have not worked on surface vehicles yet but I know there is one strange behavior that makes all things that are on the surface rotate slowly. I will try to fix it as I am implementing some stuff now.

Hedjet, 2024-12-02:
Hi, I'm working right now on a large surface vehicle in Stonefish and it works fine except for the mentioned behavior that makes the vehicle have a constant angular rotation in Z... I tried many options in the environment and vehicle physics with no success. Is there any possible solution or a future fix?

closed 2025-03-03 by patrykcieslak