patrykcieslak/stonefish #34: Is there a way to add the sensor suite from Gazebo to Stonefish?

tjsos, 2023-04-25:
It appears to be the case that there is a limited suite of sensors in [Stonefish](https://stonefish.readthedocs.io/en/latest/sensors.html). It would be nice to have a way to simply plug and play those sensors found in the [Gazebo simulator](https://classic.gazebosim.org/tutorials?tut=ros_gzplugins#Tutorial:UsingGazebopluginswithROS) in this simulated engine also.

Those Gazebo sensors work like plugins to the URDF file. They go about with a <plugin> tag and an `.so` file whereas the ones in stonefish are enabled via a <sensor> tag and type parameter. If there could exist a bridge between the two, the amount of sensors one can use in Stonefish increases.

patrykcieslak, 2023-04-26:
Honestly Gazebo and Stonefish have little in common, especially in terms of tight integration of custom graphics engine and Bullet Physics. I don't see any simple way to bridge things between the two. 

New sensors can be created just by subclassing and then you can use them by creating the scenario programmatically. Developing a simulator using the Stonefish library is simple and then in your subclass of the SimulationManager you can use the custom sensors you developed and even combine with parsing from the scenario files.

BTW which sensors are you missing? 
I see it much easier to just develop them then put time in some bridging with other software.

tjsos, 2023-04-26:
Interesting!
Thanks for the information. I was looking at the implementation of a [Velodyne LiDAR](https://classic.gazebosim.org/tutorials?tut=guided_i1&cat=#Intro) sensor into the sim environment.

closed 2023-06-08 by tjsos