patrykcieslak/stonefish #35: [Integration] Struggling to build the example

rbrigden, 2023-05-28:
**Describe the bug**
Really appreciate this project! So far I've build and run the tests, and those work great. Currently struggling to build the example simulator outside of the `stonefish` repo assuming that I've install `stonefish` to my system with `sudo make install`.

**To Reproduce**

My setup:

```
/MyWorkspace/
  /example
    /main.cpp
    /MySimulationManager.cc
    /MySimulationManager.h
 CMakeLists.txt
```

```
#  CMakeLists.txt
add_executable(demo demo/main.cpp demo/MySimulationManager.cpp)
```

My steps:

```
cd MyRepo/demo
mkdir build
cd build
cmake ..
make
```

Output:
```
In file included from /MyWorkspace/demo/main.cpp:1:
/usr/local/include/Stonefish/core/GraphicalSimulationApp.h:29:10: fatal error: graphics/OpenGLDataStructs.h: No such file or directory
   29 | #include "graphics/OpenGLDataStructs.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/demo.dir/build.make:76: CMakeFiles/demo.dir/demo/main.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/demo.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

**Expected behavior**

I expect this to build successfully and the system includes seem to work fine, but the relative includes fail to resolve. If I add an `include_target_directories`, I get linking errors.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

patrykcieslak, 2023-05-29:
Hello!
Your CMakeLists.txt has to include finding the library in the system and linking it to the executable. 
I am using the new syntax of CMake in which it should look something like this (does not require separate definition for includes):
`find_package(Stonefish REQUIRED 1.3.0)`
`add_executable(demo demo/main.cpp demo/MySimulationManager.cpp)`
`target_link_libraries(demo PUBLIC Stonefish::Stonefish)`
Let me know if this was enough, cheers!

ggcarvalho, 2024-09-04:
Hi there,

I've followed the installation procedure, but I'm currently unable to run the examples in the Tests/ directory. I'm not sure what I might be missing or doing wrong.

Any assistance or guidance would be greatly appreciated!

Thank you.

closed 2025-03-03 by patrykcieslak