patrykcieslak/stonefish #38: How to load Apriltags into the simulation?

altyi, 2023-09-19:
Hi, I want to use Apriltags in the simulation but are not sure what are some ways for loading them in? I have tried loading a static object with a custom looks using the Apriltag image as a texture. The Apriltag is loaded but are repeated over the entire surface.

altyi, 2024-06-05:
I was able to add in the apriltag as a texture for graphical materials ([looks](https://stonefish.readthedocs.io/en/latest/materials.html#looks)):

In the looks scenario xacro file, I added tags in one by one:
` <look name="apriltag0" gray="0.1" roughness="0.1" texture="Apriltag File Location"/> `  

Then, I added them into my world scenario file, an exmaple: 
`<static name="apriltag00" type="box">
        <dimensions xyz="0.20 0.20 0.0001"/>
        <material name="Steel"/>
        <look name="apriltag0" uv_mode="1"/>
        <world_transform xyz="3.25 -1.0 0.5" rpy="0.0 1.5708 3.1415"/>
    </static>
`
However, I have only tried this method for a one sided Apriltag, since this method applies the Apriltag to all surface of the object, it might not work if you need something like a cube.

devinhunsberger, 2024-06-05:
> I was able to add in the apriltag as a texture for graphical materials ([looks](https://stonefish.readthedocs.io/en/latest/materials.html#looks)):
> 
> In the looks scenario xacro file, I added tags in one by one: `<look name="apriltag0" gray="0.1" roughness="0.1" texture="Apriltag File Location"/>`
> 
> Then, I added them into my world scenario file, an exmaple: `<static name="apriltag0" type="box"> <dimensions xyz="0.20 0.20 0.0001"/> <material name="Steel"/> <look name="apriltag0" uv_mode="1"/> <world_transform xyz="3.25 -1.0 0.5" rpy="0.0 1.5708 3.1415"/> </static> `
> 
> However, I have only tried this method for a one sided Apriltag, since this method applies the Apriltag as texture to all surface of the object, it might not work if you need something like a cube.

Yeah, I have been having issues attempting to overlay some textures I made in blender for some of my team's 3D models (a red buoy for instance). I can only get it work with a box though and I would like to replicate my actual environment as much as possible. And having the ability to have separate colors on the objects is pretty important for us. I could use multiple 3d files and just have them together but that seems like alot of extra work that shouldn't exist.

devinhunsberger, 2024-06-06:
To answer your question on AprilTags, you can set the different uv_modes and the scaling, depending on if you have a singular Apriltag or a whole rectangular prism texture image. 

```lane definition has one special functionality. It is possible to scale the automatically generated texture
coordinates, to tile the textures associated with the look. In the XML syntax the <look> tag has to be augmented to
include attribute uv_scale="#.#" and in the C++ code the scale can be passed as the last argument in the object
constructor.```

```Box definition has one special functionality. It is possible to choose from 3 automatically generated texture
coordinate schemes: scheme 0 (default) assumes that the texture is in a cubemap format and applies it to the box faces
accordingly, scheme 1 applies the whole texture to each face of the box, and scheme 2 tiles the whole texture along
each of the box faces, based on face dimensions. In the XML syntax the <look> tag has to be augmented to include
attribute uv_mode="#" and in the C++ code the mode can be passed as the last argument in the object constructor.```

So depending on your AprilTag image you may need to select a different UV mode or scale to your box "looks".

closed 2025-03-03 by patrykcieslak