patrykcieslak/stonefish #42: btTransform basis and origin use different coordinate systems?

jrtcppv, 2024-01-18:
I built a very simple example using the girona500 model, it is included in the scenario as follows:
```
  <include file="girona500auv_console.scn">
    <arg name="robot_name" value="GIRONA500"/>
    <arg name="robot_position" value="5.0 3.0 10.0"/>
  </include>
```

At the start of the simulation I am calling `getRobot` on the simulation then `getTransform` on the robot to get a `btTransform` object back. I am confused about the content of the transform; the origin shows something close to `x=5, y=3, z=10` as I would expect based on my definition, however the basis I am getting something closer to:
```
[0.0, -1.0, 0.0]
[0.0, 0.0, -1.0]
[1.0, 0.0, 0.0]
```
I would expect with `rpy` set to all zeros to get something close to identity for the basis, however the basis appears to be the bullet physics coordinate system within an NED world frame. Why is the origin in NED, but the basis is in the bullet physics frame (Y up, Z forward, X left)? Or are they both in bullet coordinates and I am misinterpreting the config file?

If this is what is expected for a btTransform I can just transform the basis to NED and leave the origin as is, but wanted to check if I am interpreting this correctly. Thank you!

xgamingman, 2024-01-23:
i didn't explore much about the simulator yet, but I will try to give some advice:
1- try to see the coordinate frames of the robot in another program since the stonefish shows the 'frames' of the inertias
2- after doing step 1, if you still don't get the correct rotation matrix, try using the transpose (I had this issue with webots readings).

I will update this comment as soon as I start working on my project

jrtcppv, 2024-01-23:
I tried retrieving the sensor from the robot with `getSensor` and then calling `getSensorFrame` on that to retrieve its `btTransform`, and it gave me the identity matrix as expected. In addition, applying rotations in the config file on the robot via the `world_transform` also gave me expected results. I would be interested in understanding what exactly `Robot::getTransform` is returning as opposed to `Sensor::getSensorFrame` but for now I will close the issue. Using a sensor reference frame is more correct anyway for my application.

@xgamingman thanks for the advice. I have both modified the underwater unit test and printed out the transforms and also I am developing Python bindings for this library with SWIG in a fork. I am working on a reinforcement learning application and using RLLib and gymnasium, so needed a Python interface.

closed 2024-01-23 by jrtcppv