patrykcieslak/stonefish #46: about Current function

seunghh1313, 2024-03-29:
I have experienced and am grateful for many things through your simulation.

While using the simulation, I had a question. There is a function that allows you to add water velocity to the environment, so I added water velocity to the ocean tag in the scenario file environment, but this does not seem to work. I need help with this part.
Thank you.
[image]

Michele1996, 2024-04-02:
Hello @seunghh1313 , I think the currents are just disabled by default. You can use a service to enable them

seunghh1313, 2024-04-02:
Thank you very much I solved this issue!

closed 2024-04-02 by seunghh1313