patrykcieslak/stonefish #52: Problem with Thrusters while building Test Directory

derickcoder44, 2024-07-22:
**Describe the bug**
I'm trying out stonefish for the first time but when I try to build the test directory I keep running into an error involving "no matching call to Thruster".

**To Reproduce**
```
mkdir build
cd build
cmake -DBUILD_TESTS=ON ..
make -j8
```
**Output from make:**
```
Consolidate compiler generated dependencies of target Stonefish_test
[ 90%] Built target Stonefish_test
Consolidate compiler generated dependencies of target FlyingTest
Consolidate compiler generated dependencies of target FloatingTest
Consolidate compiler generated dependencies of target ConsoleTest
Consolidate compiler generated dependencies of target JointsTest
Consolidate compiler generated dependencies of target FallingTest
Consolidate compiler generated dependencies of target FluidDynamicsTest
Consolidate compiler generated dependencies of target UnderwaterTest
Consolidate compiler generated dependencies of target SlidingTest
[ 92%] Built target FlyingTest
[ 92%] Building CXX object Tests/CMakeFiles/FloatingTest.dir/FloatingTest/FloatingTestManager.cpp.o
[ 93%] Built target ConsoleTest
[ 95%] Built target FluidDynamicsTest
[ 95%] Built target JointsTest
[ 96%] Built target FallingTest
[ 98%] Built target SlidingTest
[ 98%] Building CXX object Tests/CMakeFiles/UnderwaterTest.dir/UnderwaterTest/UnderwaterTestManager.cpp.o
/home/beyonce/stonefish/Tests/FloatingTest/FloatingTestManager.cpp: In member function ‘virtual void FloatingTestManager::BuildScenario()’:
/home/beyonce/stonefish/Tests/FloatingTest/FloatingTestManager.cpp:71:113: error: no matching function for call to ‘sf::Thruster::Thruster(const char [9], sf::Polyhedron*&, double, std::pair<double, double>, double, double, bool)’
   71 |     sf::Thruster* thrust = new sf::Thruster("Thruster", prop, 0.3, std::make_pair(0.48, 0.48), 0.05, 500.0, true);
      |                                                                                                                 ^
In file included from /home/beyonce/stonefish/Tests/FloatingTest/FloatingTestManager.cpp:29:
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:50:3: note: candidate: ‘sf::Thruster::Thruster(std::string, sf::SolidEntity*, std::shared_ptr<sf::RotorDynamics>, std::shared_ptr<sf::ThrustModel>, sf::Scalar, bool, sf::Scalar, bool, bool)’
   50 |   Thruster(std::string uniqueName, SolidEntity* propeller,
      |   ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:51:55: note:   no known conversion for argument 3 from ‘double’ to ‘std::shared_ptr<sf::RotorDynamics>’
   51 |                        std::shared_ptr<RotorDynamics> rotorDynamics,
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note: candidate: ‘sf::Thruster::Thruster(const sf::Thruster&)’
   35 | class Thruster : public LinkActuator
      |       ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note:   candidate expects 1 argument, 7 provided
make[2]: *** [Tests/CMakeFiles/FloatingTest.dir/build.make:90: Tests/CMakeFiles/FloatingTest.dir/FloatingTest/FloatingTestManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:192: Tests/CMakeFiles/FloatingTest.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp: In member function ‘virtual void UnderwaterTestManager::BuildScenario()’:
/home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp:217:120: error: no matching function for call to ‘sf::Thruster::Thruster(const char [13], sf::Polyhedron*&, double, std::pair<double, double>, double, double, bool)’
  217 |     sf::Thruster* thSway = new sf::Thruster("ThrusterSway", prop1, 0.18, std::make_pair(0.48, 0.48), 0.05, 1000.0, true);
      |                                                                                                                        ^
In file included from /home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp:45:
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:50:3: note: candidate: ‘sf::Thruster::Thruster(std::string, sf::SolidEntity*, std::shared_ptr<sf::RotorDynamics>, std::shared_ptr<sf::ThrustModel>, sf::Scalar, bool, sf::Scalar, bool, bool)’
   50 |   Thruster(std::string uniqueName, SolidEntity* propeller,
      |   ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:51:55: note:   no known conversion for argument 3 from ‘double’ to ‘std::shared_ptr<sf::RotorDynamics>’
   51 |                        std::shared_ptr<RotorDynamics> rotorDynamics,
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note: candidate: ‘sf::Thruster::Thruster(const sf::Thruster&)’
   35 | class Thruster : public LinkActuator
      |       ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note:   candidate expects 1 argument, 7 provided
/home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp:218:127: error: no matching function for call to ‘sf::Thruster::Thruster(const char [18], sf::Polyhedron*&, double, std::pair<double, double>, double, double, bool)’
  218 |     sf::Thruster* thSurgeP = new sf::Thruster("ThrusterSurgePort", prop2, 0.18, std::make_pair(0.88, 0.48), 0.05, 1000.0, true);
      |                                                                                                                               ^
In file included from /home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp:45:
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:50:3: note: candidate: ‘sf::Thruster::Thruster(std::string, sf::SolidEntity*, std::shared_ptr<sf::RotorDynamics>, std::shared_ptr<sf::ThrustModel>, sf::Scalar, bool, sf::Scalar, bool, bool)’
   50 |   Thruster(std::string uniqueName, SolidEntity* propeller,
      |   ^~~~~~~~
[... 33 lines cut ...]
      |       ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note:   candidate expects 1 argument, 7 provided
/home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp:221:126: error: no matching function for call to ‘sf::Thruster::Thruster(const char [17], sf::Polyhedron*&, double, std::pair<double, double>, double, double, bool)’
  221 |     sf::Thruster* thHeaveB = new sf::Thruster("ThrusterHeaveBow", prop5, 0.18, std::make_pair(0.48, 0.48), 0.05, 1000.0, true);
      |                                                                                                                              ^
In file included from /home/beyonce/stonefish/Tests/UnderwaterTest/UnderwaterTestManager.cpp:45:
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:50:3: note: candidate: ‘sf::Thruster::Thruster(std::string, sf::SolidEntity*, std::shared_ptr<sf::RotorDynamics>, std::shared_ptr<sf::ThrustModel>, sf::Scalar, bool, sf::Scalar, bool, bool)’
   50 |   Thruster(std::string uniqueName, SolidEntity* propeller,
      |   ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:51:55: note:   no known conversion for argument 3 from ‘double’ to ‘std::shared_ptr<sf::RotorDynamics>’
   51 |                        std::shared_ptr<RotorDynamics> rotorDynamics,
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note: candidate: ‘sf::Thruster::Thruster(const sf::Thruster&)’
   35 | class Thruster : public LinkActuator
      |       ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note:   candidate expects 1 argument, 7 provided
make[2]: *** [Tests/CMakeFiles/UnderwaterTest.dir/build.make:104: Tests/CMakeFiles/UnderwaterTest.dir/UnderwaterTest/UnderwaterTestManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:296: Tests/CMakeFiles/UnderwaterTest.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

**Desktop:**
 - OS: Ubuntu 22.0.4 LTS

I'm new to the stonefish library so I'm not sure whether I misunderstood the build instructions or my environment is broken. Some help would be appreciated. Thanks in advance!

patrykcieslak, 2024-07-23:
This should be now fixed. I have changed the way thrusters are defined and forgot to update the example. Thanks for pointing it out!

derickcoder44, 2024-07-23:
Hi, Thanks for the fix! I think it fixed some of the errors but I still get these errors when I compile the tests. 

```
/home/beyonce/stonefish/Tests/FloatingTest/FloatingTestManager.cpp: In member function ‘virtual void FloatingTestManager::BuildScenario()’:
/home/beyonce/stonefish/Tests/FloatingTest/FloatingTestManager.cpp:71:113: error: no matching function for call to ‘sf::Thruster::Thruster(const char [9], sf::Polyhedron*&, double, std::pair<double, double>, double, double, bool)’
   71 |     sf::Thruster* thrust = new sf::Thruster("Thruster", prop, 0.3, std::make_pair(0.48, 0.48), 0.05, 500.0, true);
      |                                                                                                                 ^
In file included from /home/beyonce/stonefish/Tests/FloatingTest/FloatingTestManager.cpp:29:
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:50:3: note: candidate: ‘sf::Thruster::Thruster(std::string, sf::SolidEntity*, std::shared_ptr<sf::RotorDynamics>, std::shared_ptr<sf::ThrustModel>, sf::Scalar, bool, sf::Scalar, bool, bool)’
   50 |   Thruster(std::string uniqueName, SolidEntity* propeller,
      |   ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:51:55: note:   no known conversion for argument 3 from ‘double’ to ‘std::shared_ptr<sf::RotorDynamics>’
   51 |                        std::shared_ptr<RotorDynamics> rotorDynamics,
      |                        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note: candidate: ‘sf::Thruster::Thruster(const sf::Thruster&)’
   35 | class Thruster : public LinkActuator
      |       ^~~~~~~~
/home/beyonce/stonefish/Library/include/actuators/Thruster.h:35:7: note:   candidate expects 1 argument, 7 provided
make[2]: *** [Tests/CMakeFiles/FloatingTest.dir/build.make:90: Tests/CMakeFiles/FloatingTest.dir/FloatingTest/FloatingTestManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:192: Tests/CMakeFiles/FloatingTest.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

patrykcieslak, 2024-07-26:
Okey, I realised one other example was not compiling. Now everything should be fixed.

derickcoder44, 2024-07-29:
Yup, works great! Thanks for the Help!

closed 2024-07-29 by derickcoder44