patrykcieslak/stonefish #53: Failed to Compile Shaders

derickcoder44, 2024-07-29:
**Describe the bug**
I'm Getting a "Failed to Compile Shaders Error" when I run open the simulation in any water based environment. The simulation runs but I'm experiencing weird rendering errors with the water surface. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**To Recreate**
```
./FluidDynamicsTest &> fluid_test_out.txt
```
[fluid_test_out.txt](https://github.com/user-attachments/files/16417469/fluid_test_out.txt)

[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**

 - OS: Ubuntu 22.04
 - OpenGL Version: 4.6 (Compatibility Profile)
 - Graphics Card: AMD Radeon rx 5700 xt
 - CPU: AMD Ryzen 7 3700x 8-core processor × 16

Michele1996, 2024-11-26:
Hello, do you still have this issue?

fjiwrogno, 2025-02-11:
Meets the same problem

Michele1996, 2025-02-11:
@fjiwrogno do you have an AMD graphics card?

fjiwrogno, 2025-02-13:
@Michele1996 No, I use nvidia graphics card.

Michele1996, 2025-02-13:
and if you don't see it in PRIME profile (it happens to me) do: 

```bash
apt install nvidia-prime
```
And then set nvidia in performance mode:
```bash

fjiwrogno, 2025-02-13:
@derickcoder44 @Michele1996 Thanks for your help! 
My OpenGL is literally using the graphics built in the CPU rather than NVIDIA graphics card. 
After setting the NVIDIA to be working in Performance Mode, it works great!
[image]

closed 2025-03-03 by patrykcieslak