patrykcieslak/stonefish #59: Use stonefish for reinforcement learning

fjiwrogno, 2025-02-24:
Frst of all thanks for your development about this wonderful simulator!

Just read the relevant arxiv paper "_Stonefish: Supporting Machine Learning Research in Marine Robotics_", in the paper you mention that 

**"In [29], the authors integrated OpenAI Gym [30] with Stonefish, using a ROS interface to connect them. However, communication through ROS has been proven to slow down the training time. To overcome this issue, we exposed the data from Stonefish to be directly accessible from OpenAI Gym using Python bindings. "**

Could you provide more details about this function about how we use OpenAI Gym with python bindings to access the data from Stonefish? 

 It will be really helpful for using rl for marine robots. Thanks!

Michele1996, 2025-02-28:
Hello @fjiwrogno ,  yes give me a few days and I can push this stuff

fjiwrogno, 2025-03-01:
> Hello [@fjiwrogno](https://github.com/fjiwrogno) , yes give me a few days and I can push this stuff

Thanks for your reply, look forward to this function!

closed 2025-03-01 by fjiwrogno