patrykcieslak/stonefish #63: Possible Error in drag calculation in v1.4

Project-Evoris, 2025-03-31:
**Describe the bug**
Error in drag calculation in v1.4. 

Unsure of the cause of the error, but velocity of the vehicle in the y direction is less than in the x direction. Same result in the underwater test example, and another example using a different vehicle.

**To Reproduce**
Run the underwater test example with no current, set surge thrusters to max, the vehicle reaches an approximate velocity of 0.78 m/s.

Rotate the orientation of the vessel to face in the y axis direction, with the same thruster configuration the vehicle only reaches approximately 0.48m/s.

**Desktop (please complete the following information):**
 - OS: ubuntu 24.04

Project-Evoris, 2025-03-31:
After further testing, the issue is specific to robots defined using compound bodies. When using a single model dynamic body the behavior appears correct.

patrykcieslak, 2025-06-09:
Hello!
Thank very much you for reporting it. 
I have localised the bug very easily thanks to your observations. 
Now it is fixed.
Cheers,
Patryk

closed 2025-06-09 by patrykcieslak